collin80/GEVCU #96: PWM outputs

collin80, 2015-12-16:
GEVCU5 boards are capable of PWM on the outputs but there is currently no support for this functionality in software. It would be nice if it were possible to set up a PWM map on the GUI interface that allows the end user to calibrate the PWM output to how their gauge works.

jrickard, 2015-12-16:
The PWM output has a problem I think.  Maybe I'm wrong.  Get me
straightened out.  With this low side switching isn't ON off and OFF is on
and so forth?  So no output is a solid 12v and as we increase the pulse
width the PWM output would go DOWN instead of up.  This is a little
counterintuitive.

Jack Rickard

jrickard, 2015-12-16:
You should have a truckload full of shit by now I guess?

I spent the entire day with your father.  We rather hit it off.  He may be
me.  Or I may be him.  We had much in common in almost all directions.  He
doesn't smoke of course and is quite trim, but beyond that....

I sent a Leaf motor and inverter.  He was rather agonizing over the risk of
whether you could do it or not.  I assured him that you could easily now
that you have a Leaf to capture from, but I would supply it in any event.
I'm sure it will be more work setting up the test bench than it is reverse
engineering the CAN commands.  Could be wrong.  But there's a lot of Leaf
stuff floating around out there already and I'm not seeing much trickery.

I would like to be able to just buy these from the Nissan dealers, add a
GEVCU, and sell them.  I guess they are supposedly rated at 90kw rather
than the 100kw of the UQM, but other than that, appears to be very similar.

I kind of put the squeeze on to talk your father into doing a plastic or
fiberglass reproduction of a 190SL in electric with a Tesla drive unit in
it.  I think you'd do extremely well with that. I'd probably have to have
one...

Jack Rickard

collin80, 2015-12-16:
Yes, you are correct. The outputs are low side and switch ground so PWM is completely backward of what you would normally expect. High duty cycle corresponds to a lower average voltage and lower duty cycle raises the average voltage. So, for voltage sensitive gauges the whole thing is backward. For gauges that count pulses it can work as normal as you would still get the correct number of pulses, just with reverse polarity. I think that would still work OK.

A somewhat bigger issue is how irritating it is if you need a frequency controlled gauge. Controlling the frequency of the PWM outputs isn't a lot of fun. It's much simpler when you can just control the duty cycle.

neuweiler, 2017-04-01:
In ArduinoDue branch SystemIO::handleTick() (around line 100) I use PWM to output the estimated SOC to a DOUT. It is reversed due to the MOSFET. Works pretty fine for my digital fuel gauge with a resistor and capacitor to smoothen out the PWM to an analog signal.

collin80, 2017-04-02:
Thanks for the heads up. Yes, I can see how you've implemented it. Using analogWrite works fine but as you know it's inverted and also does not support setting the frequency. Normally you don't need to change frequency but I had an application where I had to. I've still got that code in a private branch, perhaps I'll see what it would take to bring it into mainline.