collin80/GEVCU #102: Use this on an electric car

amrithmmh, 2018-08-30:
I am working on a college project that needs a whole CAN bus system to control a car. The messages will be generated by a computer and send to the controller and then actuators . Can this project be used for my project?   note: currently the car have no ECU or CAN bus system

collin80, 2018-08-30:
Yeah, that's basically *all* it's for. ;) The whole point is to use it as the main ECU for an electric car in order to control all of the various other modules.

amrithmmh, 2018-08-31:
@collin80 I am new to this topic.
I have a question.Is there any Electronic control units that can be used to control accelerator  brake and steering ?
currently the car has only an electric accelerator pedal , brake is mechanical and steering is also mechanical