collin80/GEVCU #103: Non-Coda UQM

timlaw55, 2019-08-28:
I have seen hints in various forums that you were considering adding support for non-Coda UQM controllers.  Do you have an outline of the work that needs to be done and perhaps some hints so I could take on this project?

Thanks

collin80, 2019-08-29:
Yeah, I never did add support because I never actually had a non-Coda UQM controller so I didn't have any need to write the code. 

The basic steps should be the same as here:
https://github.com/collin80/GEVCU6/blob/master/docs/howtocreatemodule.docx

That is, copy an existing object and rename everything, add a new entry to DeviceTypes.h, add an instantiation in the ino file within the createObjects function. The system handles enabling/disabling each device automatically so you should be able to copy the existing coda control class and edit it to match the non-coda version. Then get the system to know it exists and everything should work.

timlaw55, 2019-08-30:
Thank you!  I will be working on this soon.  Do you by chance have a copy of the "UQM Diagnostic Software”?  I had a copy but lost it when my HD crashed.

Thanks again,

Tim

timlaw55, 2020-07-28:
I guess "I will be working on this soon." (from my last message) means almost a year later.  After going through the UQM CAN manual, the only difference I can see between Coda and native UQM is byte 5 of the payload is not part of the UQM spec.  I was planning to just remove that byte and give the controller a try.  I thought I would post here and ask if there were any other differences that you may know of that I need to look for.

srlake, 2021-01-20:
@timlaw55 did you end up getting this working with the non-CODA UQM inverter?