collin80/GEVCU #104: Compiling Code Throws Warnings

chadman98126, 2020-02-12:
Colin/Jack:

When compiling the GEVCU code, the latest version of the Arduino IDE (1.8.11) throws the following errors when writing text strings, logging, etc.  - several dozen actually.

obviously these are warnings and the code will still compile, but pretty cumbersome having to debug around these.  appears to be a syntax issue between C/C++. one approach might be to use 
const char* vs. char* in each of the declarations -- the issue is primarily setting commonName and when sending log messages -- there may be others, but that's as far as I've gotten through all the warnings.

your thoughts on whether this is a really BAD idea or other options - as you've been closest to the code throughout.

note: I'm loathe to suppress the warnings in the IDE, etc. for obvious reasons, but if that's the only practical option...

--------

sketch/CanBrake.cpp: In constructor 'CanBrake::CanBrake()':
sketch/CanBrake.cpp:40:13: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
  commonName = "CANBus brake";
             ^
sketch/CanBrake.cpp: In member function 'virtual void CanBrake::setup()':
sketch/CanBrake.cpp:46:71: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
  Logger::info("add device: CanBrake (id: %X, %X)", CANBRAKEPEDAL, this);

chadman98126, 2020-02-12:
note: I updated char* to const char* in device.h and device.cpp and the bulk of these warnings go away.   spelunking through the remaining 4 or 5 in GEVCU.ino

but again, not sure if these were oversights -- as the code elsewhere uses const char* -- of if these were intentional.

chadman98126, 2020-02-12:
these are the remaining errors.  not as clear to me how to resolve, but will continue to dig.

---------
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino: In function 'void initializeDevices()':
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino:230:66: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
  Logger::info("add: Heartbeat (id: %X, %X)", HEARTBEAT, heartbeat);
                                                                  ^
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino: In function 'void setup()':
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino:261:28: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
  Logger::info("TWI init ok");
                            ^
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino:263:62: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
  Logger::info("add MemCache (id: %X, %X)", MEMCACHE, memCache);
                                                              ^
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino:268:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
        Logger::info("Initializing EEPROM");
                                          ^
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino:272:60: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
           else {Logger::info("Using existing EEPROM values");}//checksum is good, read in the values stored in EEPROM
                                                            ^
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino:285:30: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
  Logger::info("SYSIO init ok");
                              ^
/Users/chad/Documents/Arduino/GEVCU/GEVCU.ino:293:29: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
  Logger::info("System Ready");

chadman98126, 2020-02-12:
again, looks to be an issue with char* declarations in logger.h and logger.cpp

now, getting very wary of making these changes...  (as Logger is used *everywhere*)

chadman98126, 2020-02-12:
and maybe only specific to Logger::info??

collin80, 2020-02-12:
It should be fine to do what you're doing. The code for GEVCU is partly kind of old and the good compiler warnings didn't trigger back then. So, we just kind of never did it right and the compiler didn't complain. Now it properly complains. It's best to make all the string constants "const" anyway as that ensures that they'll be left in FLASH instead of loaded into RAM where they needn't be.

chadman98126, 2020-02-12:
Problem is the declarations in logger.h and .cpp are tripping me up — I’m not a super coder.  they are char *variableName (vs char* variableName) and not close enough to the original code to get it to compile. After making changes to logger - all sort of things go hairy.

cgalpin, 2020-02-13:
What’s hairy?

In both logger.h and logger.cpp you should be able to change all "char *message” to “const char *message” in the method arguments without issues.

chadman98126, 2020-02-13:
I tried that and I got a raft of errors. I wasn’t sure if “const char *message” was a valid declaration.

cgalpin, 2020-02-13:
I downloaded Arduino IDE 1.8.11, and the latest GEVCU, installed the Due board and selected it and compiled. I then changed all the instances that gave warning and it compiled cleanly.

If you change the function declaration in a .h, you need to also do the same in the .cpp file

Here is there diff. 

hth
charles

$ diff -r GEVCU GEVCU-master
diff -r GEVCU/Device.cpp GEVCU-master/Device.cpp
43c43
< const char* Device::getCommonName() {
---
> char* Device::getCommonName() {
diff -r GEVCU/Device.h GEVCU-master/Device.h
60c60
< 	const char* getCommonName();
---
> 	char* getCommonName();
69c69
< 	const char *commonName;
---
> 	char *commonName;
diff -r GEVCU/GEVCU.ino GEVCU-master/GEVCU.ino
92,104c92,104
< sendWiReach((char *)"AT+iHIF=1");//Host connection set to serial port
< sendWiReach((char *)"AT+iBDRF=9");//Automatic baud rate on host serial port
< sendWiReach((char *)"AT+iRPG=secret"); //Password for iChip wbsite
< sendWiReach((char *)"AT+iWPWD=secret");//Password for our website
< sendWiReach((char *)"AT+iWST0=0");//Connection security wap/wep/wap2 to no security
< sendWiReach((char *)"AT+iWLCH=4");  //Wireless channel
< sendWiReach((char *)"AT+iWLSI=GEVCU");//SSID
< sendWiReach((char *)"AT+iWSEC=1");//IF security is used, set for WPA2-AES
< sendWiReach((char *)"AT+iSTAP=1");//Act as AP
< sendWiReach((char *)"AT+iDIP=192.168.3.10");//default ip - must be 10.x.x.x
< sendWiReach((char *)"AT+iDPSZ=8");//DHCP pool size
< sendWiReach((char *)"AT+iAWS=1");//Website on
< sendWiReach((char *)"AT+iDOWN");//Powercycle reset
---
> sendWiReach("AT+iHIF=1");//Host connection set to serial port
> sendWiReach("AT+iBDRF=9");//Automatic baud rate on host serial port
> sendWiReach("AT+iRPG=secret"); //Password for iChip wbsite
> sendWiReach("AT+iWPWD=secret");//Password for our website
> sendWiReach("AT+iWST0=0");//Connection security wap/wep/wap2 to no security
> sendWiReach("AT+iWLCH=4");  //Wireless channel
> sendWiReach("AT+iWLSI=GEVCU");//SSID
> sendWiReach("AT+iWSEC=1");//IF security is used, set for WPA2-AES
> sendWiReach("AT+iSTAP=1");//Act as AP
> sendWiReach("AT+iDIP=192.168.3.10");//default ip - must be 10.x.x.x
> sendWiReach("AT+iDPSZ=8");//DHCP pool size
> sendWiReach("AT+iAWS=1");//Website on
> sendWiReach("AT+iDOWN");//Powercycle reset
diff -r GEVCU/Logger.cpp GEVCU-master/Logger.cpp
37c37
< void Logger::debug(const char *message, ...) {
---
> void Logger::debug(char *message, ...) {
50c50
< void Logger::debug(DeviceId deviceId, const char *message, ...) {
---
> void Logger::debug(DeviceId deviceId, char *message, ...) {
63c63
< void Logger::info(const char *message, ...) {
---
> void Logger::info(char *message, ...) {
76c76
< void Logger::info(DeviceId deviceId, const char *message, ...) {
---
> void Logger::info(DeviceId deviceId, char *message, ...) {
89c89
< void Logger::warn(const char *message, ...) {
---
> void Logger::warn(char *message, ...) {
102c102
< void Logger::warn(DeviceId deviceId, const char *message, ...) {
---
> void Logger::warn(DeviceId deviceId, char *message, ...) {
115c115
< void Logger::error(const char *message, ...) {
---
> void Logger::error(char *message, ...) {
128c128
< void Logger::error(DeviceId deviceId, const char *message, ...) {
---
> void Logger::error(DeviceId deviceId, char *message, ...) {
141c141
< void Logger::console(const char *message, ...) {
---
> void Logger::console(char *message, ...) {
201c201
< void Logger::log(DeviceId deviceId, LogLevel level, const char *format, va_list args) {
---
> void Logger::log(DeviceId deviceId, LogLevel level, char *format, va_list args) {
246c246
< void Logger::logMessage(const char *format, va_list args) {
---
> void Logger::logMessage(char *format, va_list args) {
372a373,374
> 
> 
diff -r GEVCU/Logger.h GEVCU-master/Logger.h
40,48c40,48
< 	static void debug(const char *, ...);
< 	static void debug(DeviceId, const char *, ...);
< 	static void info(const char *, ...);
< 	static void info(DeviceId, const char *, ...);
< 	static void warn(const char *, ...);
< 	static void warn(DeviceId, const char *, ...);
< 	static void error(const char *, ...);
< 	static void error(DeviceId, const char *, ...);
< 	static void console(const char *, ...);
---
> 	static void debug(char *, ...);
> 	static void debug(DeviceId, char *, ...);
> 	static void info(char *, ...);
> 	static void info(DeviceId, char *, ...);
> 	static void warn(char *, ...);
> 	static void warn(DeviceId, char *, ...);
> 	static void error(char *, ...);
> 	static void error(DeviceId, char *, ...);
> 	static void console(char *, ...);
57,58c57,58
< 	static void log(DeviceId, LogLevel, const char *format, va_list);
< 	static void logMessage(const char *format, va_list args);
---
> 	static void log(DeviceId, LogLevel, char *format, va_list);
> 	static void logMessage(char *format, va_list args);
62a63,64
> 
>

chadman98126, 2020-02-13:
I will try this. Much thanks!

chadman98126, 2020-02-13:
got a subsequent similar error on ThottleDetector.cpp, but resolved in s similar way.
all the char* errors now seem to be resolved.

now getting the following:

sketch/MemCache.cpp: In member function 'boolean MemCache::Write(uint32_t, void*, uint16_t)':
sketch/MemCache.cpp:214:81: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       pages[c].data[(uint16_t)((address+count) & 0x00FF)] = *(uint8_t *)(data + count);
                                                                                 ^
sketch/MemCache.cpp: In member function 'boolean MemCache::Read(uint32_t, void*, uint16_t)':
sketch/MemCache.cpp:274:27: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
       *(uint8_t *)(data + count) = pages[c].data[(uint16_t)((address+count) & 0x00FF)];

will take a run at these next

chadman98126, 2020-02-13:
disregard the last, it appears the Arduino IDE was parsing MemCache.cpp incorrectly and missing some of the declaration.  compiling a second time, the warnings go away.