collin80/GEVCU #107: Support newer version of Arduino IDE

timlaw55, 2020-07-24:
The readme specifies the need to use Arduino IDE 1.5.4.  It will not run on OS X Catalina.  Are there specific reasons requiring this version?

collin80, 2020-07-24:
Probably not but maybe. The thing is, the source code for GEVCU is right around 5-7 years old at this point. Back then the most recent IDE version probably was 1.5.4. So, the idea, as far as I recall, was to require a fairly new IDE version so that people wouldn't try to use a super old version. The newer 1.8.X release which is current should work. There have been some architectural changes to the Arduino system over the years but the Arduino Due core that is used by GEVCU hasn't had updates in quite some time so it is likely pretty stable and just about the same as it was 5 years ago. The GEVCU6 branch that supports the newer V6 hardware was branched from this code and it has been recently compiled on 1.8.X versions and works fine. So, I have reason to believe it would work fine but I can't promise that.

cgalpin, 2020-07-24:
I can confirm it works fine on 1.8.11 on OSX - just tried it the other day

timlaw55, 2020-07-24:
I was using 1.8.x.  I added my new motor controller and got everything compiling.  When I tried to push it to GEVCU 5.2 I got this error :

    No device found on cu.usbmodem14101
    An error occurred while uploading the sketch

I know the port is good because I can interact with the GEVCU through the integrated terminal.  I tried bringing up 1.5.4 IDE version in a VM but I can't even get that to build.  Idealy I would rather use the newer version.  Thanks @cgalpin for verifying it works.  Now if I could figure out why it fails to push the code.  Do I need to crack open the case and press a button or something to get it to accept programming?  Any other suggestions for why I would get that error?

timlaw55, 2020-07-24:
After failing a number of times, I turned on verbose logging and it uploaded. :/  I show a new option for UQM.  Now on to the UQM CAN spec documents to figure out what I need to send to get the motor spinning...

closed 2020-07-24 by timlaw55